connornishijima/Pixie_Chroma #56: Be clear about chip support limitations

ladyada, 2022-04-06:
hiya it does not seem to be stated anywhere in the documentation that only ESP32, ESP8266 or Teensy 3 boards are the **only** chips supported by pixie chroma - however, this line in the code will not let it compile otherwise
https://github.com/connornishijima/Pixie_Chroma/blob/main/src/Pixie_Chroma.h#L56

can you clarify if that is correct, **only ESP32, ESP8266 or Teensy 3 boards can be used** and if so, where is that documented?

connornishijima, 2022-04-07:
Whoops, thank you! It had slipped my mind after all of the development process to add that in explicitly! (Also, wow, hi! You and others made this career possible for me.)

*Yes, Chroma only officially supports ESP8266, ESP32, and Teensy at the time of writing, due to RAM and FastLED compatibility limitations.*

There is now a snippet in the README about supported controllers, planned future support, and steps to help users add experimental support themselves.

https://github.com/connornishijima/Pixie_Chroma#supported-platforms

connornishijima, 2022-04-15:
*Quick note before closing this issue* - I've written a much better "Getting Started" guide for Pixie Chroma that Adafruit is more than welcome to link to when the time comes that they go up for sale:

https://connornishijima.github.io/Pixie_Chroma/?section=getting-started

Thank you again for your help!

closed 2022-04-15 by connornishijima

ladyada, 2022-04-16:
[image]

clicking this link does not work

connornishijima, 2022-04-16:
Oops! That's a properly written URL, but it's trying to open an external HTTPS site (github.com) in the content iFrame, which violates CORS policy.

The page you saw is rendered with JavaScript directly from getting_started.md on this repo, I'll need to add some quick code in the morning that forces the parser to generate links with a `target="_parent"` attribute to force them to open in the parent window. Good catch!

connornishijima, 2022-04-16:
Fixed in [280b825](https://github.com/connornishijima/Pixie_Chroma/commit/280b825d3577c0d8093d7a7418cafe0be9bfe489).

ladyada, 2022-04-17:
ok great!