connornishijima/Pixie_Chroma #64: cannot compile sample code

JimKruk, 2022-06-22:
I bought these from adafruit and am using the huzzah esp32 feather board and was following the tutorial here:
https://connornishijima.github.io/Pixie_Chroma/?section=getting-started

I installed the library (1.2.0) and then opened the sampled code:
Examples > Pixie_Chroma > 01_Basic > 01_Getting_Started

but when I try and compile it I get an error:
C:\Users\Jim\OneDrive\Documents\Arduino\libraries\Pixie_Chroma\src/Pixie_Chroma.h:74:138: fatal error: FastLED.h: No such file or directory

thanks

connornishijima, 2022-06-22:
Did you install the library manually by downloading the repo, or through the Arduino Library Manager? 

"FastLED" is a second library this one depends on that should have been automatically installed - if not, just search "FastLED" in the same Library Manager window and install the latest version to see if that fixes it! :)

JimKruk, 2022-06-22:
I installed pixie chroma using the arduino manager library (was following the tutorial).  I just installed FastLED in the library manager and that worked, thanks !  The getting started tutorial was very well written, perhaps someone could include that step after installing the pixie chroma library.

JimKruk, 2022-06-22:
my arduino IDE is 1.8.5, good to know, thanks.  One of the reasons I decided to purchase these is your doc is very well written, easy to follow and has lots of examples.  For a beginner that is extremely helpful and encouraging to see someone take the time to publish good documentation, thanks !

connornishijima, 2022-06-22:
Thanks Jim! I'll add the Arduino IDE version note to the Getting Started Guide for others later today.

connornishijima, 2022-06-23:
Notes about minimum IDE version support and workaround notes for older versions added to getting_started.md in commit [f237a12](https://github.com/connornishijima/Pixie_Chroma/commit/f237a12af17912b89888e0cc71dc8f26aa637ef7)!

closed 2022-06-23 by connornishijima

Shy, 2024-03-07:
Hey folks, I'm also having this issue. The libraries are being auto installed to FastLED 3.60. I'm getting a lot of errors related to FastLED. 

This is using the current version of the Arduino IDE. Do you know if I need an older version of FastLed to make this work? I can't get the sample code or even the minimal example to run. 

```
In file included from /Users/shy/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/esp_common/include/esp_macros.h:13,
                 from /Users/shy/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:76,
                 from /Users/shy/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/freertos/FreeRTOS-Kernel/include/freertos/portable.h:59,
                 from /Users/shy/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:71,
                 from /Users/shy/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/Arduino.h:33,
                 from /private/var/folders/1p/hymkpcx11hj3hbnspzt0n84m0000gn/T/arduino/sketches/56EF4E59F7462CE253620D3C60494F15/sketch/01_Standard.ino.cpp:1:
/Users/shy/Documents/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h: In member function 'void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::init()':
/Users/shy/Documents/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:234:25: error: 'GPIO_PIN_MUX_REG' was not declared in this scope; did you mean 'GPIO_PIN19_REG'?
  234 |         PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[DATA_PIN], PIN_FUNC_GPIO);
      |                         ^~~~~~~~~~~~~~~~
/Users/shy/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/esp_common/include/esp_assert.h:24:38: note: in definition of macro 'TRY_STATIC_ASSERT'
   24 |             if (__builtin_constant_p(CONDITION) && !(CONDITION)) {          \
      |                                      ^~~~~~~~~
/Users/shy/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/soc/esp32/include/soc/soc.h:34:57: note: in expansion of macro 'IS_DPORT_REG'
   34 | #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
      |                                                         ^~~~~~~~~~~~
/Users/shy/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/soc/esp32/include/soc/soc.h:83:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
   83 |             ASSERT_IF_DPORT_REG((_r), REG_SET_FIELD);                                                                  \
      |             ^~~~~~~~~~~~~~~~~~~
/Users/shy/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/soc/esp32/include/soc/io_mux_reg.h:83:46: note: in expansion of macro 'REG_SET_FIELD'
   83 | #define PIN_FUNC_SELECT(PIN_NAME, FUNC)      REG_SET_FIELD(PIN_NAME, MCU_SEL, FUNC)
      |                                              ^~~~~~~~~~~~~
/Users/shy/Documents/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:234:9: note: in expansion of macro 'PIN_FUNC_SELECT'
  234 |         PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[DATA_PIN], PIN_FUNC_GPIO);
      |         ^~~~~~~~~~~~~~~
In file included from /Users/shy/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/xtensa-esp32-elf/sys-include/sys/reent.h:503,
                 from /Users/shy/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/newlib/platform_include/sys/reent.h:9,
                 from /Users/shy/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/xtensa-esp32-elf/sys-include/stdio.h:60,
                 from /Users/shy/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0-alpha3/cores/esp32/Arduino.h:27:
/Users/shy/Documents/Arduino/libraries/FastLED/src/platforms/esp/32/clockless_i2s_esp32.h:234:25: error: 'GPIO_PIN_MUX_REG' was not declared in this scope; did you mean 'GPIO_PIN19_REG'?
  234 |         PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[DATA_PIN], PIN_FUNC_GPIO);
      |                         ^~~~~~~~~~~~~~~~
/Users/shy/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/soc/esp32/include/soc/soc.h:34:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
   34 | #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
      |                                      ^~~~~~~~~~~~~~~~~
/Users/shy/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32/include/soc/esp32/include/soc/soc.h:34:57: note: in expansion of macro 'IS_DPORT_REG'
   34 | #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
      |                                                         ^~~~~~~~~~~~
[... cut ...]
/Users/shy/Documents/Arduino/libraries/Pixie_Chroma/src/utility/../utility/../pixie_chroma_internal.h:107:32: error: 'void PixieChroma::println(long unsigned int)' cannot be overloaded with 'void PixieChroma::println(uint32_t)'
  107 |                 /*|*/     void println( long unsigned int input ); // pretty much the same as uint32_t, but Arduino is stupid
      |                                ^~~~~~~
/Users/shy/Documents/Arduino/libraries/Pixie_Chroma/src/utility/../utility/../pixie_chroma_internal.h:105:28: note: previous declaration 'void PixieChroma::println(uint32_t)'
  105 |                 /*|*/ void println( uint32_t input );
      |                            ^~~~~~~

```