connornishijima/Pixie_Chroma #70: FastLED interrupt is panicking on esp32

iamJoeTaylor, 2023-01-21:
**Describe the bug**
I had the same issue on a TinyPico and a Feather esp32. Calling `pix.begin` causes a failure in the `FastLED.show` method because Pixie Chroma is calling `noInterrupts`. FastLED is waiting for interrupts to fire but the interrupts have been disabled causing it to hang and fail with watchdog.

I had this code working in April of last year, but trying to flash it recently caused this issue. If I comment out the `noInterrupts` and `interrupts` calls in Pixie Chroma lib it is able to print to the LED screens.

**To Reproduce**
Steps to reproduce the behavior (using the Getting Started)
1. `pix.begin( DATA_PIN, PIXIES_X, PIXIES_Y );` with an esp32.
2. Build and upload with platform io
3. See error:
```
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40091487: vListInsert at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/list.c:183
WARNING Decoded 0x400901a7: vTaskPlaceOnEventList at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c:3566
WARNING Decoded 0x4008f52a: xQueueSemaphoreTake at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c:1688
WARNING Decoded 0x400f0181: ClocklessController<5, 60, 150, 90, (EOrder)66, 0, false, 5>::showPixels(PixelController<(EOrder)66, 1, 4294967295u>&) at /Users/joetaylor/Development/esphome-pixie-chroma/.esphome/build/pixie-chroma/.piolibdeps/pixie-chroma/FastLED/platforms/esp/32/clockless_i2s_esp32.h:582
 (inlined by) ClocklessController<5, 60, 150, 90, (EOrder)66, 0, false, 5>::showPixels(PixelController<(EOrder)66, 1, 4294967295u>&) at /Users/joetaylor/Development/esphome-pixie-chroma/.esphome/build/pixie-chroma/.piolibdeps/pixie-chroma/FastLED/platforms/esp/32/clockless_i2s_esp32.h:546
WARNING Decoded 0x400e5529: CPixelLEDController<(EOrder)66, 1, 4294967295u>::show(CRGB const*, int, CRGB) at /Users/joetaylor/Development/esphome-pixie-chroma/.esphome/build/pixie-chroma/.piolibdeps/pixie-chroma/FastLED/controller.h:408
WARNING Decoded 0x400d2771: CLEDController::showLeds(unsigned char) at /Users/joetaylor/Development/esphome-pixie-chroma/.esphome/build/pixie-chroma/.piolibdeps/pixie-chroma/FastLED/controller.h:90
 (inlined by) CFastLED::show(unsigned char) at /Users/joetaylor/Development/esphome-pixie-chroma/.esphome/build/pixie-chroma/.piolibdeps/pixie-chroma/FastLED/FastLED.cpp:59
WARNING Decoded 0x400e2ef1: CFastLED::show() at /Users/joetaylor/Development/esphome-pixie-chroma/.esphome/build/pixie-chroma/.piolibdeps/pixie-chroma/FastLED/FastLED.h:500
 (inlined by) PixieChroma::show() at /Users/joetaylor/Development/esphome-pixie-chroma/.esphome/build/pixie-chroma/.piolibdeps/pixie-chroma/Pixie_Chroma/src/pixie_chroma_internal.cpp:2419
WARNING Decoded 0x400e2f25: PixieChroma::set_animation(void (*)(PixieChroma*, float)) at /Users/joetaylor/Development/esphome-pixie-chroma/.esphome/build/pixie-chroma/.piolibdeps/pixie-chroma/Pixie_Chroma/src/pixie_chroma_internal.cpp:346
WARNING Decoded 0x400e391c: PixieChroma::begin(unsigned char, unsigned char, unsigned char) at /Users/joetaylor/Development/esphome-pixie-chroma/.esphome/build/pixie-chroma/.piolibdeps/pixie-chroma/Pixie_Chroma/src/pixie_chroma_internal.cpp:105
```

(Note: this error is from my espHome module but the same error and trace happens from a fresh arduino platform io build as well)

**Expected behavior**
Begin should no panic at Watchdog causing any programs to fail.

**Desktop (please complete the following information):**
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6) 
Dependency Graph
|-- Pixie_Chroma @ 1.2.0
|   |-- Ticker @ 2.0.0
|   |-- FastLED @ 3.5.0
|   |   |-- SPI @ 2.0.0

romkey, 2023-06-14:
I can confirm; I see the same problem with both a regular ESP32 and an ESP32-S2. Removing the interrupt lockout solves the issue for me as well.

romkey, 2023-06-14:
This problem happens with both PlatformIO and Arduino IDE.

Arduino framework 2.0.4 is the last version that doesn't crash this way. More recent versions do. Not sure what the change was that brings out the problem.

 @connornishijima do you remember why you were disabling interrupts for this section of code?