connornishijima/Pixie_Chroma #73: Cannot convert CRGB to uint8_t

tomasro, 2023-11-13:
**Describe the bug**
Error while compiling code:
...\Arduino\libraries\Pixie_Chroma\src\pixie_chroma_internal.cpp: In member function 'void PixieChroma::color_blur_x(fract8)':
...\Arduino\libraries\Pixie_Chroma\src\pixie_chroma_internal.cpp:1874:25: error: cannot convert 'CRGB' to 'uint8_t' {aka 'unsigned char'} in assignment
carryover = part;
                   ^~~~

tomasro, 2023-11-15:
Also happens with IDE 1.8, and the only thing I had in my test code was #include "Pixie_chroma.h"

tomasro, 2023-11-15:
Turns out this error is caused because of FastLED 3.6. Reverting to 3.5 clears the error, but now there is a CPU panic when running pix.begin()

AlainVagner, 2024-03-23:
I had a similar issue, here is what I did to solve it. I am using an esp32.
I rolled back to the following versions:
 
- Arduino SDK 1.8.19
- esp32 SDK version 3.0.2 in the board manager (this version appears in a screenshot on the "getting started" page). 
- FastLED v3.5

strain10280, 2024-06-09:
esp32 SDK versão 2.0.3

strain10280, 2024-06-09:
ESP32 SDK 2.0.9 works fine

andyshinn, 2025-01-08:
Sorry for the whitespace changes but see https://github.com/andyshinn/Pixie_Chroma/commit/a51597a74aed737822ae0c20c333d1cb8d6bd625 for a fix.